WouterTuinstra/Homeassistant-Growatt-Local-Modbus #1: Mismatched data on larger inverters

lordblc, 2023-03-05:
Hi
Love the integration. It was just what i was looking for, but.. I have 4 strings in my inverter model and i suspect that might have done something with the setup as i get weird lines.. Some data seems to fit correctly, others are WAY off or just not present..
What would you need me to do to get you the needed information to make the changes?

WouterTuinstra, 2023-03-07:
Hi
During my investigation into the growatt inverters I already noticed that different inverter models would use different structures to store data values in the modbus registers. Although it would be relatively to add a new register structure it would require a small redesign for maintainability and ease of use.

The main item that I would require currently is knowing which protocol version your inverter is using potentially I can derive this also from the inverter model as well.

From another repo I found the following:
- The older inverters apparently use Protocol v3.05 from year 2013. -> Currently implemented
- The newer inverters apparently use protocol v1.05 from year 2018.
- There is also a new protocol version v1.24 from 2020.

lordblc, 2023-03-08:
Thank you for quick reply. :)
Hmm.. I have no way to determine what protocol they use (not a programmer), but the model no for my specific model is MID 17KTL3-XL
It was installed last year, so i would be guessing it's of the newest sort, but then again.. I have no way to tell.. 
I will gladly test it for you though.

WouterTuinstra, 2023-03-17:
I have created a [new branch](https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/tree/multi_protocol) and #2 where I made a significant redesign and implemented a new protocol version 1.20

If you could have a look and test if this protocol decodes the information correctly that would be great.
The config flow tries to detect based on modbus version device information.
If you get none please check your full home-assistant logbook it will show for both protocols the related information.

I would be mainly interested in the modbus version and device type number that is being decoded from the device.

lordblc, 2023-03-18:
Yep, that solved it. :)
Thank you!

closed 2023-03-18 by lordblc

WouterTuinstra, 2023-03-18:
That is great to hear I have merged the pull request.