WouterTuinstra/Homeassistant-Growatt-Local-Modbus #5: unable to connect

monisoja, 2023-07-14:
hello, I have growatt local running now some time for 2 invertors. after a update from raspbarry pi 3 to 4 i can not connect any new invertors any more. 1 is running and 1 is failing. when i try to reconnect it again i get "unknow error occurred".
When i remove the one that is working and reinstal everything i get this error on both of them. The only way to get 1 working back is to do a restore of a backup.

I there a way to change the modbus adres without reinstaling the invertor

WouterTuinstra, 2023-07-16:
Hello, 
There is currently no option to change this from the front end of HA. This would require a Options Flow.
What you can do is go to the folder of home assistant and open `.storage/core.config_entries`

Search for `growatt_local` and update the address value in the configuration. 

Are both inverters connected on a single serial port or do you have two serial ports connected to your raspberry pi? 
Did you also check the home assistant logs?

monisoja, 2023-07-16:
Hello Wouter,

THANX!!!!. this did solve my problem.  Now it is connected again. 
I checked my home assistant logs. thats way i wanted to change the port adres. Why i can not connect any new ( removed) invertors any more i dont know. i only get the "unknown error occurred"   
I have the 2 inverters on saparete modbus to usb dongels. This was already the best option for the rasp.3. I changed to the rasp.4 because the port numbers kept changing after every restart. so i had to reinstal the invertors everytime i did a restart. ( i have 3 modbus connections. and it going to be more). with the rasp.4 it is not a problem any more. 

Many thanx for the nice addon!

closed 2023-07-16 by monisoja