WouterTuinstra/Homeassistant-Growatt-Local-Modbus #6: Support Growatt SPF 5000 ES

Didou30, 2023-08-01:
Hello,

 does this repository support SPF 5000 ES and the usb of the inverter must be connected to the HA server?

Thanks for your response.

WouterTuinstra, 2023-08-02:
Hi,

There is currently no full support for an off grid inverter mainly because of not knowing which information would be relevant and would be missing.

if you are using a usb to serial device and running HA in docker yes you will have to add access to that device for the HA container.

Didou30, 2023-08-02:
hello,

thanks for your response, all the information would be relevant to see the modbus register, but I don't know how to implement them!
[Growatt OffGrid SPF5000 Modbus RS485 RTU Protocol.pdf](https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/files/12238183/Growatt.OffGrid.SPF5000.Modbus.RS485.RTU.Protocol.pdf)

Didou30, 2023-08-02:
This code in YAML for Esphome content all informations:
`text_sensor:
  - platform: template
    name: "${friendly_name} Status"
    icon: mdi:eye
    entity_category: diagnostic
    lambda: |-
      if ((id(status).state) == 1) {
        return {"Normal"};
      } else if ((id(status).state) == 0)  {
        return {"Standby"};
      } else if ((id(status).state) == 2)  {
        return {"Discharge"};
      } else if ((id(status).state) == 3)  {
        return {"Fault"};
      } else if ((id(status).state) == 4)  {
        return {"Flash"};
      } else if ((id(status).state) == 5)  {
        return {"PV Charging"};
      } else if ((id(status).state) == 6)  {
        return {"AC Charging"};
      } else if ((id(status).state) == 7)  {
        return {"Combined Charging"};
      } else if ((id(status).state) == 8)  {
        return {"Combined Charging & Bypass"};
      } else if ((id(status).state) == 9)  {
        return {"PV Charging & Bypass"};
      } else if ((id(status).state) == 10)  {
        return {"AC Charging & Bypass"};
      } else if ((id(status).state) == 11)  {
        return {"Bypass"};
      } else if (id(status).state == 12)  {
        return {"PV Charge and Discharge"};
      } else {
        return {"Unknown"};
      }

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s
    
  - platform: modbus_controller
    address: 0
    register_type: "read"
    internal: true
    accuracy_decimals: 0
    id: status
    
#### temp ###
 
  - platform: modbus_controller
    name: "${friendly_name} Inverter Temperature"
    address: 25
    register_type: "read"
    unit_of_measurement: °C
    device_class: temperature
    entity_category: diagnostic
    state_class: measurement
    icon: mdi:thermometer
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
    - multiply: 0.1
    
  - platform: modbus_controller
    name: "${friendly_name} PV Temperature"
    address: 32
    register_type: "read"
    unit_of_measurement: °C
[... 479 lines cut ...]
      "45,6": 456
      "45,8": 458
      "46,0": 460
      "46,2": 462
    #   ...       ...    check the range in the manual
 
  - platform: modbus_controller
    name: "${friendly_name} Mains to battery operating point"
    address: 95
    value_type: U_WORD
    optionsmap: 
      "48,0": 480
      "48,2": 482
      "48,4": 484
      "48,6": 486
      "49,0": 490
      "49,2": 492   
      "49,4": 494   
      "49,6": 496   
      "49,8": 498  
      "50,0": 500   
      "51,0": 510   
      "52,0": 520  
    #   ...       ...    check the range in the manual
    `

WouterTuinstra, 2023-08-03:
I have made an initial concept for support of the Offgrid SPF inverter. Please be aware that it's has not been tested yet.
This is not including the select sensors from above only readable sensor values.

https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/tree/offgrid-inverter

Didou30, 2023-08-04:
Hello,
Thanks you, i will test

WouterTuinstra, 2023-08-10:
Hello,

I have tested the integration and found an issue that wouldn't allow you to use the new protocol this has been fixed.
Now I would like to get a verification that the items are getting decoded correctly and optionally suggestions for the default naming convention used by the integration.

Didou30, 2023-08-11:
Hello,

i will tested next week, thanks

verjoz, 2023-09-04:
Hi WouterTuinstra,

I also have an offgrid inverter (same as Didou30 you have - SPF 5000 ES) and after many strugle with the other way (using growatt servers) I am really happy that you made a local version! :)

I am started to use the offgrid branch, added the custom repo, started the config_flow.
I chose serial, and entered the serial port (/dev/serial/by-id/usb-04e2_1410-if00) and the rest (left default).
And I get "Unkown error occured".

I am not running hacs in docker.
The ls -l /dev/ttyACM0 shows root and audio.
I added the hassio user to the audio group, not sure what user is executing the serial port reading in hacs.
(And ha is new to me, so not sure if I added the right user (hassio). :(

Could you help me what to check, where can I find log file to see what is failing?

Your quick response would be very much apprecieted :)

Thank you,

verjoz, 2023-09-04:
I found the logs but did not understand why is this failing:

^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 181, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 296, in async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 392, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/growatt_local/config_flow.py", line 287, in async_step_serial
    await server.close()
  File "/config/custom_components/growatt_local/API/growatt.py", line 74, in close
    await self.client.close()
TypeError: object NoneType can't be used in 'await' expression

Didou30, 2023-09-05:
Hello,

not tested, i hope this week..

WouterTuinstra, 2023-09-05:
Hello,

In order to access the device within a docker container it has to be added when starting the container.
So you would have to add the device argument when creating the hass container:
`--device="/dev/ttyACM0"`

I'm not sure why the client would become None at this point

verjoz, 2023-09-16:
> <WouterTuinstra/Homeassistant-Growatt-Local-Modbus/issues/6/1707133102@
> github.com>

Hi WouterTuinstra,

Thank you for your quick response. I tried to figure out how to do that,
but unfortunately, I did not succeed.
Could you help how to do this?

I started with a flashed SD card on a RPI4 (HA OS as I can remember), so it
is not me who is running it in a container.
I checked by ssh-ing into the PI that in docker it is running these
containers:
[image: image.png]
But I don't know who is running this, I could not find a compose.yaml to
edit the parameters (--device=/dev/ttyACM0).
(the HA OS has a boot time "supervisor" as I tried to

I tried to stop it and rerun it, with the below command, but then all my
settings are lost (luckily I renamed the original one as a backup :))

 docker run -d \
  --name homeassistant \
  --privileged \
  --restart=unless-stopped \
  -e TZ=Europe/Budapest \
  -v /config \
  --network=host \
  --device="/dev/ttyACM0" \
  ghcr.io/home-assistant/home-assistant:stable

Your help would be very much appreciated. :)

WouterTuinstra ***@***.***> ezt írta (időpont: 2023. szept.
5., K, 20:37):

WouterTuinstra, 2023-09-29:
Hi @verjoz,

I'm afraid that I can't help with that because I'm not running HA OS and do not know where it's original docker configration to startup Home Assistant would be located.

You could explore an alternative methode to run a serial to network converter tool.

verjoz, 2023-09-29:
> <WouterTuinstra/Homeassistant-Growatt-Local-Modbus/issues/6/1740614584@

Ok, thank you, will try.

WouterTuinstra ***@***.***> ezt írta (időpont: 2023. szept.
29., P, 11:48):

Saentist, 2024-02-20:
> hello,
>
> thanks for your response, all the information would be relevant to see the modbus register, but I don't know how to implement them! [Growatt OffGrid SPF5000 Modbus RS485 RTU [...]

Newer version of RTU protocol [OffGrid-Modbus-RS485RS232-RTU-Protocol-V0.17-20220622](https://www.amosplanet.org/wp-content/uploads/2023/01/OffGrid-Modbus-RS485RS232-RTU-Protocol-V0.17-20220622.pdf)

closed 2024-12-31 by WouterTuinstra