WouterTuinstra/Homeassistant-Growatt-Local-Modbus #7: Support Growatt MIN 6000TL XH

l4m4re, 2023-08-31:
I have a Growatt MIN 6000TL XH coupled to an ARK 10 kWh battery system with HVC 60050-A1 BMS and a [backup box](https://www.stralendgroen.nl/product/growatt-backup-box/) in order to have the possibility to go off-grid when needed. In order for this to work, an RS-485 power meter is installed in order to measure the power to/from the grid, which is coupled to the syscom port of the inverter, which is why I can't use that port for the integration.

So, I decided to try a usb-to-rs485 converter both on the inverter as well as on the RPI, on which I'm running HA OS. The first ones I tried didn't work, which is because the inverter only works with a CH340, as can be read [here](https://github.com/otti/Growatt_ShineWiFi-S):

> I tested several ESP8266-boards with builtin USB-Serial converters so far only boards with CH340 do work (CP21XX and FTDI chips do not work). 

Now I got it working with the offgrid-inverter branch, which I [forked](https://github.com/l4m4re/Homeassistant-Growatt-Local-Modbus/tree/offgrid-inverter) and installed as described in the readme, but there are some issues.

The first is that the status of the inverter goes to "no response" quite often. It may be because I used the cheapest usb converters I could find, but it may also be because the inverter talks to the battery manager as well as the power meter over modbus as well, albeit over separate 9600 bps interfaces. In any case, it seems a good idea to implement retries and I was thinking of modyfing _async_update_data in __init__.py as well as read_holding_registers and read_input_registers in growatt.py. 

The next issue is that there is a total power entry for input 1, but not for input 2, which I suspect is caused by a "no response" when scanning for available registers. Since I intend to add more registers (such as the number of cycles my batteries have made), I would like to know how to re-scan.

Further, I had to set the protocol version manually. I already checked the protocol version register before and IIRC it said 3.05, even though it appears to work quite well with v1.20.  

This is what's currently in my core.config_entries:

`
      {                                                                                                                                                   
        "entry_id": "1db4177128478ebcd9662c260a8cb347",                                                                                                   
        "version": 1,                                                                                                                                     
        "domain": "growatt_local",                                                                                                                        
        "title": "Growatt MIN 6000TL XH",                                                                                                                 
        "data": {                                                                                                                                         
          "communication_layer": "serial",                                                                                                                
          "port": "/dev/ttyUSB0",                                                                                                                         
          "baudrate": 115200,                                                                                                                             
          "stopbits": 1.0,                                                                                                                                
          "parity": "None",                                                                                                                               
          "bytesize": 8.0,                                                                                                                                
          "address": 1,                                                                                                                                   
          "serial_number": "SNL0CGV020",                                                                                                                  
          "firmware": "AL1.0ZAba\u001d",                                                                                                                  
          "name": "Growatt ",                                                                                                                             
          "model": "MIN 6000",                                                                                                                            
          "type": "hybride_120",                                                                                                                          
          "dc_string": 2.0,                                                                                                                               
          "ac_phases": 1.0,                                                                                                                               
          "scan_interval": 60,                                                                                                                            
          "power_scan_enabled": true,                                                                                                                     
          "power_scan_interval": 20                                                                                                                       
        },                                                                                                                                                
        "options": {},                                                                                                                                    
        "pref_disable_new_entities": false,                                                                                                               
        "pref_disable_polling": false,                                                                                                                    
        "source": "user",                                                                                                                                 
        "unique_id": "SNL0CGV020",                                                                                                                        
        "disabled_by": null                                                                                                                               
      },
 `

WouterTuinstra, 2023-09-04:
I have had some initial problems with getting the interfacing working as well. I found out that the CH340 that I coincidental have as well worked on a windows device but on a SBC running linux it didn't work correctly the data was malformed. I set up a different converter for my SBC and it worked fine.

You could check whether it's data range is the problem or just the general data return on any device but running a part of the API only. The following code example (using offgrid-inverter-branch) can be run from within the API test functionality without requiring to run Home Assistant.

```
import asyncio
from growatt import GrowattSerial, GrowattDevice, DeviceTypes
from utils import RegisterKeys

growatt = GrowattDevice(
    GrowattSerial("/dev/ttyUSB0"),
    DeviceTypes.HYBRIDE_120,
    1
)

async def main():
    await growatt.connect()

    result = await growatt.update(RegisterKeys(input=set(range(0, 30))))

    print(result)

asyncio.run(main())
```
Feel free to implement retries but be aware that only could be advise to use in certain conditions:
* When running relatively high update rates it should be limited amount
* Solar inverters only will startup later than when then sun is up and default have a time window where status is: no_response
* It should not run infinite amount of retries to get all register values.

_It may also be that the time-out is too short. Growatt specifies 850ms._ -Potentially yes, but I'm not effected by it I can send the message directly after I received the last one

_I already checked the protocol version register before and IIRC it said 3.05_ -My feeling says that the same value is returned for all growatt devices and that it is a library version or so and not based on the register format protocol they designed. v1.20 would be this protocol design: https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf

l4m4re, 2023-09-04:
Thanks a lot for that example code, I'll see if I can get that working on my old laptop.  I've ordered a couple of new CH340 usb sticks from two different vendors. The ones I have do not have any markings on the chip, so I guess those are unstable clones. At first, I could easily swap the stock ShineWifi stick back in and that would work, but now the inverter just won't respond anymore even after just a few modbus read attempts and I have to reset the inverter before the stock stick works again. So, once the new converters arrive I'll see if I can scan registers and inputs based on your example.

When looking through the code, I saw there already are retries in the GrowattNetwork init, but not in GrowattSerial init. I tried editing the ones in GrowattNetwork, but now I realize these are not used for the serial interface. :( 

The timeout for serial is set to 3 seconds, so that's not the problem. In any case, AsyncModbusSerialClient should also support the retries option, since that comes from [ModbusBaseClient](https://pymodbus.readthedocs.io/en/v3.0.2/source/library/pymodbus.client.html#pymodbus.client.base.ModbusBaseClient).

> v1.20 would be this protocol design: https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf

In that document, it says: V1.05 2018-04-19, while it only has 55 pages and when you search for "xh" in this one, you find nothing.

I found a v1.20 version here that has 65 pages:

https://blog.moospirit.org/post/2022/growatt/growatt-inverter-modbus-rtu-protocol-ii-v1-20-english.pdf

However, I also found a v1.24 version that has 85 pages here: 

https://tuxdoc.com/download/growatt-inverter-modbus-rtu-protocol-ii-v1-24-english-new_pdf

In these two, there is quite some info about register ranges of various inverters, including the X(H) range.

l4m4re, 2023-09-09:
The new sticks arrived and I have been able to make a scan using modbus-scanner, which I forked and edited. The result can be found in this file:

https://github.com/l4m4re/python-modbus-scanner/blob/main/scan3.txt

I only printed the registers/inputs that had a non-zero value, since the inverter responded to lots of registers/inputs with a value of 0 (see scan1.txt, which is incomplete).

Further, it seems that these declarations are wrong in inverter_120.py
 
```
    GrowattDeviceRegisters(
        name=ATTR_OUTPUT_REACTIVE_POWER, register=58, value_type=float, length=2,
    ),
    GrowattDeviceRegisters(
        name=ATTR_OUTPUT_REACTIVE_ENERGY_TODAY, register=60, value_type=float, length=2,
    ),
    GrowattDeviceRegisters(
        name=ATTR_OUTPUT_REACTIVE_ENERGY_TOTAL, register=62, value_type=float, length=2,
    ),
```
These registers are already declared further up as part of other length 2 values, so they can't be correct.

The next one, register 65, is declared twice.

closed 2024-12-31 by WouterTuinstra