WouterTuinstra/Homeassistant-Growatt-Local-Modbus #9: Step by step -  after installing Growatt Modbus

RVis78, 2023-10-08:
I am not sure, but hopefully this is a workaround to get Growatt sensors working - because of the problem Growatt integration sensors unavailable, after growatt changed their request policy.

I've installed the Growatt Modbus, but is there a step by step tutorial?

At the first step of the configuration I am totally lost.

I don't know which comm. method toe use: Serial/TCP/UDP ?? How do i know which one to choose?
[image]

And the second step: ??
[image]

I haven't a clue which specific info i need to fill in, or where to find the specific info.

please can you help me, so my Energy dashboard will functioning as it should.

WouterTuinstra, 2023-10-08:
Hi @RVis78,

If you expected that you found an easy replacement for the your server integration you are wrong. 
I have specified the steps as simple as possible within the configuration possibilities of your inverter
First up would be what type of inverter do you have and how are you going to connect to it. 
If the answer is via network or wifi the answer would be by default TCP
However if you have a Growatt shine adaptor where you want to connect to you should be aware that this has never been tested and might not work.
There are many ways to get your local assigned IPv4 address just use a software tool to scan your network and find the IP address assigned to to your growatt inverter

Then a last item that would be on a next page is the Protocol version but again that would be dependent on the type of the inverter.

RVis78, 2023-10-09:
Hi Wouter,Thanks for your reply. I have a shine link adapter. I'll let you know when the connection is successful!Met vriendelijke groet,Remco Visser

closed 2024-01-07 by WouterTuinstra