WouterTuinstra/Homeassistant-Growatt-Local-Modbus #13: Unable to make connection Growatt MIN 3000TL-X

vtecfreak, 2023-10-22:
Currently I am not be able to communicate to the Growatt inverter via Modbus.
I have a Waveshare RS485 To Eth(B) connected to the Growatt MIN 3000TL-X to the Sys-com connector.
The network connection to the Waveshare is working because the "Link" LED turns to blue which means the connection is opened successfully, so far so good. The "ACT" LED is blinking when it tries to communicate with the interverter via Modbus, but the following error messages appear:
"Device didn't respond".
"Device address is likely wrong".

I have checked the address of the inverter and it was set to 58. However I have changed it to 1, but still no luck.

Full details:
```
Communication layer: tcp
IP: <internal ip-address of Waveshare>
Port: 502
Modbus device address: 1
Modbus Framer: Modbus TCP
```

I am not sure if you are familiar with the Waveshare module, but the following is set:
```
Baudrate: 9600
Data bits: 8
Parity: None
Stop bits: 1
Flow control: none
Modbus gateway type: Simple Modbus TCP to RTU
```

Do I miss something?

Saentist, 2024-02-09:
Elfin EW11 settings with work
[image]
[image]

vtecfreak, 2024-02-10:
> Unfortunately I'm not familiar waveshare module. Did you find a solution?

Thanks for your reply. But no, unfortunately not. But I have to admit I did not spend much time recently to this issue because of other projects I am working on.

vtecfreak, 2024-02-10:
> Elfin EW11 settings with work [...]

Thanks, that looks interesting! To make sure: this setup with the Elfin EW11 is actually working with this plugin in HA?

Saentist, 2024-02-10:
> Thanks, that looks interesting! To make sure: this setup with the Elfin EW11 is actually working with this plugin in HA?

Waveshare have same settings just different UI.

vtecfreak, 2024-02-11:
> Waveshare have same settings just different UI.

It is worth a try. However I cannot recognise all the settings of the Elfin compared with Waveshare, but I will take a closer look.

Saentist, 2024-02-11:
> It is worth a try. However I cannot recognise all the settings of the Elfin compared with Waveshare, but I will take a closer look.

Post what you have to compare
basic is to have modbus to inverter set to 9600/8/n

vtecfreak, 2024-02-11:
> > It is worth a try. However I cannot recognise all the settings of the Elfin compared with Waveshare, but I will take a closer look.
> 
> Post what you have to compare basic is to have modbus to inverter set to 9600/8/n

Currently these are the settings as in VirCom:

Standard settings:
[image]

Advanced settings:
[image]

vtecfreak, 2024-02-11:
After comparing your settings and the settings in WaveShare I noticed the Flow Control was set to ```Half duplex``` in your configuration. In WaveShare it was set to ```None```. So I have changed this to ```CTS\RTS``` (= half duplex as I can remember) and now I am be able to make a connection with this plug-in in Home Assistant!
Details will follow to tomorrow, because it is now dark at the moment. Status in HA is now ```Offline```.

Saentist, 2024-02-11:
> Currently these are the settings as in VirCom:

What are other option of "Transfer Protocol" is there RTU
why not use web interface of device
[image]

vtecfreak, 2024-02-12:
> What are other option of "Transfer Protocol" is there RTU why not use web interface of device [...]

Well, because of a stupid reason: I cannot remember the password I have set in the webinterface. I will have to reset the device back to factory settings. But I wanted to try out it first with the new setting, therefor I used the VirCom.
However, both interfaces can be used.
In VirCom I only the option ```Simple Modbus TCP to RTU```, regarding to ```RTU```.

But what I can remember of the webui, it is not that extensive like yours. Let me get back to you as soon a I have the webui running back again.

vtecfreak, 2024-02-12:
> Well, because of a stupid reason: I cannot remember the password I have set in the webinterface. I will have to reset the device back to factory settings. But I wanted to try out it first with the [...]

Herewith the settings via the WebUI. It does not much differ from VirCom and there a not a lot of options compared with your interface of WaveShare.

[image]