WouterTuinstra/Homeassistant-Growatt-Local-Modbus #14: RTU setting not saved

brupje, 2023-10-28:
This was driving me crazy for a while, but I temporarily found a workaround. I have a Growatt Shinelan, where I uploaded the default firmware and thus removed the Grotwatt cloud. It uses RTU and I could add the device fine in growatt_local, but upon loading entities it would throw errors like ValueError: Sensor sensor.total_energy_produced has device class 'energy', state class 'total_increasing' unit 'kWh' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unavailable' (<class 'str'>)

Stepping through the code I added some debug information in growatt.py:

`class GrowattNetwork(GrowattModbusBase):
    def __init__(
        self,
        network_type: str,
        host: str,
        port: int = 502,
        frame: str = '',
        timeout: int = Defaults.Timeout,
        retries: int = Defaults.Retries,
    ) -> None:
        """Initialize Network Growatt."""

        _LOGGER.info("Reconnecting to growatt %s - %s", network_type.lower() ,frame.lower()  )
        if network_type.lower() == "tcp":
            if frame.lower() == 'rtu':
                _LOGGER.info("Connecting using TCP-RTU to %s:%d with timeout %s",host,port,timeout)
                self.client = AsyncModbusTcpClient(
                    host,
                    port if port else Defaults.TcpPort,
                    framer=ModbusRtuFramer,
                    timeout=timeout,
                    retries=retries,`

and it didn't go through the RTU part as frame appears to be empty. Adding frame = 'rtu' to the top of the function fixes my problem, but it seems selecting RTU in the config flow is not saved or something, but I am not sure how to debug or fix that.

WouterTuinstra, 2023-10-28:
Hi,

Checked out the initialization point of the `GrowattNetwork` class in async_setup_entry and found that it didn't configure the framer option. The config flow configuration does save this information in your config so you should be good now.

closed 2023-10-28 by WouterTuinstra

brupje, 2023-10-29:
Yes, that seems to have fixed it. Thanks!