WouterTuinstra/Homeassistant-Growatt-Local-Modbus #27: [BUG] HA 2024.02.0 integration not start

Saentist, 2024-02-08:
After update Integration do not start

```
Core 2024.2.0
Supervisor 2024.01.1
Operating System 11.5
Frontend 20240207.0
```

```
Invalid config
The following integrations and platforms could not be set up:
growatt_local Show logs
```

```py
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 08:49:19 ч. (1 occurrences)
Last logged: 08:49:19 ч.

Setup failed for custom integration 'growatt_local': Unable to import component: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 827, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 56, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 17, in <module>
    from pymodbus.client import ModbusBaseClient
ImportError: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py)

```

WouterTuinstra, 2024-02-09:
Which version is currently installed of pymodbus because there is a range of releases where the `ModbusBaseClient` in not included in the pymodbus/client/\_\_init__.py this has been fixed in the most recent release 3.6.4 
see also following commit: https://github.com/pymodbus-dev/pymodbus/commit/66af5b67058605c81854e872b0ff3f612dcb278e

Saentist, 2024-02-09:
Any example how to get version?
I use Home Assistant HAOS in ESXI server
https://www.home-assistant.io/installation/alternative 
everything is preinstalled

Saentist, 2024-02-11:
```
Support for pyModbus below 3.5.2 has been dropped, for HA installations older than 2023.9.0
Integration 2023.09.3 is the last supported version
```

No idea what is current version.

Nick-Innes, 2024-02-18:
I'm suffering the same error message, and the integration failing to start once I upgraded from 2024.1.x to 2024.2.1, and the issue persists after upgrading to 2024.2.2, and removing and reinstalling the integration via HACS.

Saentist, 2024-02-18:
How you remove it it's impossible, delete not working?

Nick-Innes, 2024-02-18:
I removed the folder below custom_components and then told HACS to ignore that it was configured. I then restarted home assistant a couple of times and reinstalled using HACS, only to find I was exactly where I'd started.
I'd been hoping that 2024.2.2 or later would maybe update pyModbus or similar, but no such luck.
I've nothing else using Modbus, except the GivTCP add-on, but that's in a different container.

WouterTuinstra, 2024-02-18:
I updated the `manifest.json` file with the requirements to use the latest version of pymodbus this should automatically solve the issue because HA will check the required packages.

Saentist, 2024-02-18:
Still have this message

```
Invalid config
The following integrations and platforms could not be set up:

[growatt_local] ([Show logs])
Please check your config and [logs].
```
```py
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 17:03:58 ч. (1 occurrences)
Last logged: 17:03:58 ч.

Setup failed for custom integration 'growatt_local': Unable to import component: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 827, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 56, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 17, in <module>
    from pymodbus.client import ModbusBaseClient
ImportError: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py)

```

WouterTuinstra, 2024-02-18:
Have you updated the integration because I have no issues. At restart it should state the following:
Attempting install of pymodbus>=3.6.4

WouterTuinstra, 2024-02-18:
I don't know why it would not update on your side best to otherwise do it manual but I don't know how to get access to Home Assistant in an ESXI environment

WouterTuinstra, 2024-02-18:
Use the "Advanced SSH & Web Terminal" and execute the following commands:
`pip show pymodbus`
`pip install pymodbus==3.6.4`

Saentist, 2024-02-18:
I guess i need to be in some docker crap before call this 
[image]

Strange before it was working
 
Second command do something
[image]

And after restart it work, but sun is not enough to see values.

WouterTuinstra, 2024-02-18:
Strange that initially not even recognized the package.

closed 2024-02-18 by WouterTuinstra

Nick-Innes, 2024-02-19:
Thank you! I updated the custom component yesterday and when the sun returned today everything worked perfectly again!

Saentist, 2024-02-24:
@WouterTuinstra just after update to [Release 2024.2.3](https://www.home-assistant.io/blog/2024/02/07/release-20242/#release-202423---february-22)
immediately block integration and need  to install `pymodbus` again.

other integration with use pymodbus have this:
```yaml
"requirements": ["pymodbus>=3.5.2"],
"version": "2024.02.6"
```

WouterTuinstra, 2024-02-25:
I'm not sure why it's not installing `pymodbus` but that would be a problem in HA itself and not in this intergration.

Saentist, 2024-02-25:
> I'm not sure why it's not installing `pymodbus` but that would be a problem in HA itself and not in this intergration.

Solax integration use it but not sure how, without need to install it as in this case.

https://github.com/home-assistant/core/tree/dev/homeassistant/components/modbus
```
# homeassistant.components.modbus
pymodbus==3.6.4
```

I see some string 
```py
from pymodbus.exceptions import ModbusException
import pytest
```
not sure is this connected

Saentist, 2024-02-27:
@WouterTuinstra 
Again update again failed to start
reload missing, need full reload of HA
[image]
need to be something as this 
[image]
after install pymodbus
[image]

Saentist, 2024-02-29:
@WouterTuinstra 
today's update show pymodbus as installed but there is some difference.

```py
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 15:09:35 ч. (1 occurrences)
Last logged: 15:09:35 ч.

Setup failed for custom integration 'growatt_local': Unable to import component: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 827, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 56, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 17, in <module>
    from pymodbus.client import ModbusBaseClient
ImportError: cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.12/site-packages/pymodbus/client/__init__.py) 
```

but pymodbus is in different location

[image]

[image]