WouterTuinstra/Homeassistant-Growatt-Local-Modbus #28: Missing meter

frankiep, 2024-02-12:
Hi man!
i'm setting up your integration to connect my MIN TL-XH inverter. 
In my system i have the inverter (modbus address 2) and an Eastron SDM120 meter (modbus address 1). The meter is connected to ports  5/6 (according to attached inverter manual information) and on server.growatt.com i see inverter and meter information.

I have connected the rs485/tcp adapter to ports 3/4 (according to attached inverter manual information).

Now setting up your integration...if i insert modbus address 2 i got the inverter working well with all informations, but if i insert modbus address 1 it does not find the meter.

How can i do to getting meter works?

Thanks 

[image]

Saentist, 2024-02-12:
Address of inverter is 1 do not confuse with other interface.
Some features are not implemented, or is in progress of implementation.

frankiep, 2024-02-12:
> Address of inverter is 1 do not confuse with other interface. Some features are not implemented, or is in progress of implementation.

in my setup inverter has modbus address 2 in fact the integration detect it and show all values.

So, at the moment, your integration can show only data from inverter? So is it useless for hybrid inverter to get info from battery and meter?

WouterTuinstra, 2024-02-18:
Yes, the main focus of this integration is based upon my own inverter however I did make updates for a hybrid inverter but that is still in a separate branch here https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/tree/storage-offgrid-hybrid-inverters. Main reason for it still being separate is nobody is able to test and give feedback on potentially missing items.
That is why https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/pull/18 is still open.

I do have to mention that the Eastron SDM120 meter will fall out of the scope of this project because it focuses the Growatt Inverters only.