WouterTuinstra/Homeassistant-Growatt-Local-Modbus #29: Changing modbus address on SPF5000TL HVM-P?

RoganDawes, 2024-02-17:
Hi folks, I am trying to support 5 parallel Growatt SPF5000TL HVM-P inverters (arranged 2-2-1 on three phases). However, I am so far unable to figure out how to change the modbus address of the inverters from the default "1". I have searched for days, but the closest I have found is an entry in the modbus register documentation that describes "knocking". This apparently applies to an earlier Growatt inverter. My inverters appear to be manufactured around 2020.

I have seen the entry in the register description for the address, which returns the expected "1". However, trying to change the address results in the inverter completely failing to respond to queries afterwards. (For reference, I am currently talking to it via the USB connector - RS232 over D+/D-, but intend to talk RS485 once I manage to solve the addressing problem). I have also seen a Reddit post that suggests that the inverter responds to queries to the new address with responses from the old address, until after it has been cold-rebooted. I tried this with no change, still no response from the inverter. To get it to respond again, I had to factory reset the inverter, and then manually reconfigure it to match the rest of them. At which point, the address was reset to 1, of course.

WouterTuinstra, 2024-02-18:
I'm not sure why the inverter completely fails after changing just the address.

I will have to note that this project was designed based on having a single wire connection to a single device.
It would not be impossible to redesign but will require significant changes.

RoganDawes, 2024-02-19:
Thanks for the response.

To be honest, I'm just looking for anyone who may have success in changing the modbus address on Growatt inverters. I'm not necessarily planning to use this particular project once I manage to do so! So no need to rearchitect with my use case in mind.