WouterTuinstra/Homeassistant-Growatt-Local-Modbus #33: Sensors keep last value when inverter is offline

matrover, 2024-07-25:
Hi, 

I've managed to get this working on my Growatt MOD 7000TL3-X with the Waveshare RS485 TO ETH. 
Really love this, many thanks!

Question the power sensors all stay on the latest value when the sun is down. The status sensor (in HA of the inverter) also shows offline. 
[image]
[image]

As sensors values can not be edited via automations (without some tricks) could we maybe add setting the values to 0 if the status is offline for longer than X ?

WouterTuinstra, 2024-07-26:
Hi,

It's great to hear that you like the integration. I implemented the status offline in the integration and is not read from the device itself. There would be no need to implement setting power values to zero. From the images, I can see that the integration stopped earlier than the inverter stopped producing power. The normal status flow should be Normal > Waiting > Offline. 

Due to the orientation of your solar panels, you are still producing power beyond the point I hardcoded to go offline which is 10 minutes before sunset. If you would change the timedelta below it delays when the integration goes offline.
https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/blob/d294ebf9da4dcd891dc33a75da568e665b03a599/custom_components/growatt_local/__init__.py#L157

matrover, 2024-07-27:
thanks for you answer @WouterTuinstra!

will try and report back!

matrover, 2024-07-29:
Just to update, ive tried 2 evenings and see the offline status shifting but its not after the actual time that the inverter is stopping power delivery. 
So ive added some extra minutes to the delta again. lets hope it will work tonight :)

matrover, 2024-07-30:
Yes! yesterday the valua became unavailable so thats good! 

Thanks very much @WouterTuinstra

closed 2024-07-30 by matrover