WouterTuinstra/Homeassistant-Growatt-Local-Modbus #40: crazy values on HA reboot

wire67, 2024-12-27:
I am getting some crazy invalid values, it seems to be corelated to when HA is rebooting. 
It seems to affect all energy values (kWh) and some current as well (A).
Could you confirm, is the Modbus parity check or CRC implemented and enabled already?

[image]
[image]

WouterTuinstra, 2024-12-31:
I have checked the Modbus functionality. When a CRC is included in the message and an invalid message is received, it disregards it. So, it depends on how you interface it whether or not you can exclude the Modbus communication.

WouterTuinstra, 2024-12-31:
Today I have seen crazy values caused most likely by running multiple instances of HA that are communicating to the same device