WouterTuinstra/Homeassistant-Growatt-Local-Modbus #44: "Message":"Invalid handler specified"

kocur80, 2025-01-05:
Hi
After the update, the integration stopped working. When trying to make a new entry, a message like this pops up. The problem is to even delete the integration. Do you have any suggestions on what I could do?

WouterTuinstra, 2025-01-05:
Hi, After which update HA or this intergration?
To narrow it down, I would need more information about HA, which should provide a call stack if it's an error.
Or more info on which logger is raising this message

kocur80, 2025-01-05:
About 3 days ago there was an update of growatt local integration.

Rejestrator: homeassistant.setup
Źródło: setup.py:334
Pierwsze zdarzenie: 12:45:05 (1 zdarzenia)
Ostatnio zalogowany: 12:45:05 

Setup failed for custom integration 'growatt_local': Unable to import component: No module named 'pymodbus.framer.rtu_framer'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 38, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 19, in <module>
    from pymodbus.framer.rtu_framer import ModbusRtuFramer
ModuleNotFoundError: No module named 'pymodbus.framer.rtu_framer'

WouterTuinstra, 2025-01-05:
Thanks for the additional info I did notice that the pymodbus intergration has changed a quite bit since version 3.7.0 
I'm not sure if it works but I have pushed a change to that should force to use a version of the https://github.com/pymodbus-dev/pymodbus version 3.6.9 however if you have other intergrations that require newer versions of the intergration it might cause failing intergrations.

kocur80, 2025-01-05:
You're right. Now  growatt local is working but modbus integration isn't.  Modbus is more important for me.

Saentist, 2025-01-06:
some more errors

```
Logger: homeassistant.setup
Източник: setup.py:334
Възникнал за първи път: 15:45:13 (1 събития)
Последно регистриран: 15:45:13

Setup failed for custom integration 'growatt_local': Unable to import component: cannot import name 'BYTE_ORDER' from 'pymodbus.framer.base' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 38, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 19, in <module>
    from pymodbus.framer.rtu_framer import ModbusRtuFramer
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/rtu_framer.py", line 7, in <module>
    from pymodbus.framer.base import BYTE_ORDER, FRAME_HEADER, ModbusFramer
ImportError: cannot import name 'BYTE_ORDER' from 'pymodbus.framer.base' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/growatt_local/__init__.py", line 38, in <module>
    from .API.growatt import GrowattDevice, GrowattSerial, GrowattNetwork
  File "/config/custom_components/growatt_local/API/growatt.py", line 19, in <module>
    from pymodbus.framer.rtu_framer import ModbusRtuFramer
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/rtu_framer.py", line 7, in <module>
    from pymodbus.framer.base import BYTE_ORDER, FRAME_HEADER, ModbusFramer
ImportError: cannot import name 'BYTE_ORDER' from 'pymodbus.framer.base' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py)```

```
Logger: homeassistant.util.loop
Източник: util/loop.py:77
Възникнал за първи път: 15:45:13 (1 събития)
Последно регистриран: 15:45:13

Detected blocking call to import_module with args ('custom_components.growatt_local',) in /usr/src/homeassistant/homeassistant/loader.py, line 1075: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module Traceback (most recent call last): File "<frozen runpy>", line 198, in _run_module_as_main File "<frozen runpy>", line 88, in _run_code File "/usr/src/homeassistant/homeassistant/__main__.py", line 227, in <module> sys.exit(main()) File "/usr/src/homeassistant/homeassistant/__m [... cut ...]  line 165, in async_setup_component result = await _async_setup_component(hass, domain, config) File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component component = await integration.async_get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component comp = self._get_component() File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component ComponentProtocol, importlib.import_module(self.pkg_path)```

closed 2025-01-11 by WouterTuinstra