WouterTuinstra/Homeassistant-Growatt-Local-Modbus #47: Error with new version

matrover, 2025-01-12:
Hi Wouter, 

Just installed the latest version, gives me the error, any idea?  

`
Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 14:42:22 (1 occurrences)
Last logged: 14:42:22

Error setting up entry Growatt MOD 7000TL3-X for growatt_local
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/growatt_local/__init__.py", line 75, in async_setup_entry
    device_layer = GrowattNetwork(
        entry.data[CONF_LAYER],
    ...<2 lines>...
        entry.data[CONF_FRAME],
    )
  File "/config/custom_components/growatt_local/API/growatt.py", line 176, in __init__
    self.client = AsyncModbusTcpClient(
                  ~~~~~~~~~~~~~~~~~~~~^
        host,
        ^^^^^
    ...<3 lines>...
        retries=retries,
        ^^^^^^^^^^^^^^^^
    )
    ^
TypeError: AsyncModbusTcpClient.__init__() got multiple values for argument 'framer'
`

WouterTuinstra, 2025-01-12:
Hi,

I think I fixed the issue with pull request #48

closed 2025-01-12 by WouterTuinstra

matrover, 2025-01-12:
wow that was quick! yes works again, thanks very much