WouterTuinstra/Homeassistant-Growatt-Local-Modbus #49: Unable to make RTU connection „Device closed connection“

MPHHB, 2025-01-21:
Hello, 

I have a working Modbus RTU setup: CH340/341 USB serial converter with standard settings (/dev/ttyUSB0, 9600, 1,N,8) through which I can read several modbus slaves. 

I would like to expand to my inverter through this promising integration. However, I cannot get past the second setup menu and I get the message that „Device closed connection“

It also gives this message when I have the modbus setup running and reading other devices (Eastron SDM smart meters). 

I saw in the closed issues that a missing pymodbus installation my be the issue. I established that I indeed did not have pymodbus and successfully installed its 3.6.4 version. This did not resolve the issue.

Does anyone have an idea?

Hartelijke groet aan @WouterTuinstra and thank you to you all!

[image]

MPHHB, 2025-01-23:
The issue was resolved. I was able to progress with the set-up of this integration at the time that I did not have Modbus entities working. 

If you have working Modbus entities, disable them all in your Modbus YAML file. Then set-up this integration and re-enable your Modbus entities in your Modbus YAML file again. 

That fixed it for me.

WouterTuinstra, 2025-01-27:
I find it strange that existing Modbus entities caused this behavior requiring disabling them to get this integration to work.

closed 2025-01-27 by WouterTuinstra