WouterTuinstra/Homeassistant-Growatt-Local-Modbus #50: Unable to connect - error "Unknown error occured"

lordblc, 2025-02-09:
Integration stopped communicating back in January and whatever i try to do, i just get that error.

Is there anything i can test to give more detail to this?
Keep in mind i am not a programmer, so detailed instructions are needed. :)

I have checked all connections several times to no avail. i have replaced the usb plug, tried using modbus - tcp adapters.
Currently it's connected to a tcp adapter.

[image]

I have changed the modbus address on the inverter to see if that helped, hence the screenshot shows 2.
I have tried deleting the integration, reloading and reconfig, hence the screenshot.

lordblc, 2025-02-11:
Got these errors:

`
Logger: aiohttp.server
Source: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
First occurred: 20:06:34 (1 occurrences)
Last logged: 20:06:34

Error handling request from 192.168.42.82
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/growatt_local/config_flow.py", line 364, in async_step_network
    device_info = await get_device_info(server, user_input[CONF_ADDRESS])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/growatt_local/API/growatt.py", line 460, in get_device_info
    inverter_v120 = await device.get_device_info(HOLDING_REGISTERS_120, minimal_length, unit)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/growatt_local/API/growatt.py", line 95, in get_device_info
    await self.read_holding_registers(item[0], count=item[1], slave=slave)
  File "/config/custom_components/growatt_local/API/growatt.py", line 152, in read_holding_registers
    data = await self.client.read_holding_registers(start_address, count=count, slave=slave)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 155, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 0 retries, continue with next request
`

`
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/growatt_local/API/growatt.py:152
integration: growatt_local (documentation, issues)
First occurred: 20:06:34 (1 occurrences)
Last logged: 20:06:34

No response received after 0 retries, continue with next request
`

LifeofDan-EL, 2025-03-01:
Hello, I am trying to install it for the first time and it is saying this too.

[image]

lordblc, 2025-03-02:
Hi LifeofDan
I might be able to shed some light on your setting.
I ended up replacing more components than i'd care for in the process of figuring this out..
In the end, try this:
- Disconnect the USB ShineWiFi-X stick.
- Reconnect the Modbus adapter.
- Retry the integration setup.
- If this doesn't work, get your hands on a USB - RS485 device and try again. 
    (if you are in .no i can help with that as i have several of these)

Growatt apparently doesn't like talking to 2 communication devices at the same time. Meaning.. It works _sometimes_ but is never a guarantee.. If this setup gets your system going, let your provider know as this means they loose access to remotely support you.

WouterTuinstra, 2025-03-10:
Hello @LifeofDan-EL, If manually integrating using Modbus works, it's interesting that the integration doesn't work as it is using the same Modbus library. I don't know if you tried both mudbus Framer options because I think that this might be the only difference that may cause an empty response.

LifeofDan-EL, 2025-03-10:
Yeah, it's a weird issue. What's the modbus framer option? Also does this integration send command to the inverter?

WouterTuinstra, 2025-03-10:
The framer option adds a checksum to the send message when selecting RTU with TCP this checksum is not added. Which to use depends on if the used converter is dumb just converting message to the serial line or smarter supports modbus itself and adding the checksum when forwarding the message to serial line to the inverter. 

There is minimal support for sending commands but they could be added to the intergration.

LifeofDan-EL, 2025-03-18:
I see, thanks for the explanation.