WouterTuinstra/Homeassistant-Growatt-Local-Modbus #54: StrEnum was used from growatt_local - deprecated

prmfeddema, 2025-03-30:
Logger: homeassistant.backports.enum
Source: helpers/deprecation.py:222
First occurred: 09:45:05 (1 occurrences)
Last logged: 09:45:05

StrEnum was used from growatt_local, this is a deprecated alias which will be removed in HA Core 2025.5. Use enum.StrEnum instead, please report it to the author of the 'growatt_local' custom integration

WouterTuinstra, 2025-03-31:
Please check which version you are running because this has been fixed here by this commit 68370c6a8d7d59a4850be9f15c9a0c38dbc0ca3e

prmfeddema, 2025-04-18:
Hi everyone - i have been busy with work and travel hence the delay in getting this tested. I have also changed the rs 465 interface module. Will check this in the coming days and keep you up to date

prmfeddema, 2025-04-18:
Ok - i'm unable to connect to the W610 device to HA  - either with Solax or the Growatt local modbus module. W610 is going back - looking for another device.