WouterTuinstra/Homeassistant-Growatt-Local-Modbus #55: No module named 'pymodbus.framer.rtu_framer'

prmfeddema, 2025-03-30:
Logger: homeassistant.config_entries
Source: config_entries.py:3788
First occurred: 09:45:05 (1 occurrences)
Last logged: 09:45:05

Error occurred loading flow for integration growatt_local: No module named 'pymodbus.framer.rtu_framer'

WouterTuinstra, 2025-03-31:
Please check which version you are running because this has been changed to keep compatibility with pymodbus here by this commit a3ec7ef8ff6a4c0aeed07d0d870dc1f68eb73da4

typxxi, 2025-04-18:
Would be nice to close such topic if it has been solved cause I am getting into this integration and read along about topics that are no solved.
But now here are solved issues or at least without reply. 
Time consuming to read.