WouterTuinstra/Homeassistant-Growatt-Local-Modbus #62: reading model name over modbus

EvEggelen, 2025-06-08:
This plugin is able to read the model number from the inverter through modbus ( holding register 28 and 29, modbus v305 ). However the model code it retrieves ( see below ) does not correlate with the model name, meaning that is on the device ( e.g. Growatt 7000 TL3-S). The app provided by Growatt does show this name. I cannot find the model code read through modbus on the device, datasheet or other. Does anyone know how to translate this code to the real model name ? Or is there a undocumented register where to read this. My inverter is running modbus version 3.05. Does anyone have more information on this ?
 
Inverter Model: A , could be show: “T0 Q0 PF U1 M5 S1” or “00F151”
Tx=(A&0XF00000)>>20
Qx=(A&0X0F0000)>>16
Px=(A&0x00F000)>>12
Ux=(A&0x000F00)>>8
Mx=(A&0x0000F0)>>4
Sx=(A&0x00000F

    def model(registers) -> str:
    mo = (registers[0] << 16) + registers[1]
    return "T{:X} Q{:X} P{:X} U{:X} M{:X} S{:X}".format(
        (mo & 0xF00000) >> 20,
        (mo & 0x0F0000) >> 16,
        (mo & 0x00F000) >> 12,
        (mo & 0x000F00) >> 8,
        (mo & 0x0000F0) >> 4,
        (mo & 0x00000F)
    )

WouterTuinstra, 2025-06-08:
Hi,

You want to be using RTU Protocol version 3.15 when setting up this integration.
I have a similar TL3-S model.

EvEggelen, 2025-06-08:
Thanks for the response. This is not really an issue, but more like a question how to get the get the documented model number over modbus with Growatt. Do have any idea how the Growatt gets the model name in their own app ?

The model number that is read e.g. T0 Q0 PF U1 M5 S1, what is that. I cannot find it anywhere.  Is this used with a lookup table to get the real model number ? Do you have a bit for information on this ?

WouterTuinstra, 2026-11-08:
I don't know I'm not using their app. Also I don't now how to interpreter the model number.
DTC Device Type Code Does match and gives a correct number strings and phases

closed 2026-11-08 by WouterTuinstra