WouterTuinstra/Homeassistant-Growatt-Local-Modbus #68: Battery info not read properly with WIT 30K-XHU

alexsapran, 2025-07-31:
Let me start with, this is an awesome integration!

I have an inverter (WIT 30K-XHU) with a battery system of Growatt (not sure I managed to get the exact model). 

And I managed to get it working and collecting the telemetry info, all seems good, except for the related Battery info and SOC.

Happy to provide any insight and debug, but the integration did not produce any failed logs or anything to share and help with debugging this.

WouterTuinstra, 2025-07-31:
Thanks, I'm not sure if you have seen #67, but it might be that it would already improve the available Battery info.

I suspect that it still uses input registers as defined in:
https://pdfcoffee.com/growatt-inverter-modbus-rtu-protocol-ii-v1-24-english-new-pdf-free.html

It might be possible that you need to extend the current storage register set:
https://github.com/WouterTuinstra/Homeassistant-Growatt-Local-Modbus/blob/5f12c216cb16073af38716960308a56ee9d1dc32/custom_components/growatt_local/API/device_type/storage_120.py#L117C1-L117C74

And also map them accordingly on the Home Assistant side.

alexsapran, 2025-07-31:
After looking more closely the #67 it might fix my issue as well.
Once you are ok and merge it, i can give it a test and see.

closed 2025-09-06 by WouterTuinstra