WouterTuinstra/Homeassistant-Growatt-Local-Modbus #70: Configuration not accessable

andsie-tester, 2025-08-05:
Hi,

I don't get it to be configured - where is an interface for configuration?

best regards
André

andsie-tester, 2025-08-05:
I refer to " and can be configured using the UI"... 

but where to find?

WouterTuinstra, 2025-08-05:
Hi,
You can follow this: https://www.home-assistant.io/getting-started/integration/
And search for growatt_local after you installed this integration with HACS

andsie-tester, 2025-08-07:
[image]

andsie-tester, 2025-08-07:
no configuration tab like in other integrations

WouterTuinstra, 2025-08-08:
To use this integration, you will need some basic knowledge of how to communicate with devices like your Growatt device.
In my opinion, getting the required parameters is part of exploring the realm of Home Assistant, and integrating a Growatt device is not easy. 
Explaining how to get this is beyond the purpose of this integration, and is very difficult because there are many steps involved, and explaining it from my perspective is difficult through a GitHub issue or readme.

andsie-tester, 2025-08-08:
Then is your effort to maintain an integrstion not suitable for me in lack of necessary informations. 

I have to look after other solutions like grott.

My only wish was to get the growatt integration to an interval if 1 minute refresh - but there is no way to get there...